ottoragam/Tarocco #2: Servomotor info/pinout?

zougloub, 2017-05-04:
Hi,

The documentation mentions use of an HKT30 encoder and doesn't provide the pinout, but the [HKT30 docs](http://www.sei-automation.com/source/wp-content/uploads/2014/12/HKT30.pdf) mention 5 pins, and there are 4 on the actual hardware.

Could you document it?

Thanks,

ottoragam, 2017-05-04:
Sure, I'm on it.

ottoragam, 2017-05-04:
It looks like there are several versions of the encoder. Some use the HEDS-97xx style modules and others use the HEDS-91xx family.

I'll upload a motor datasheet that includes the pinout of the encoder module in a moment.

ottoragam, 2017-05-04:
Please take a look at the new servomotor datasheet, it now includes the encoder pinout.

zougloub, 2017-05-04:
Turns out (#3) the encoder really needs 5V, at least mine doesn't output anything at 3.3V.

ottoragam, 2017-05-04:
Yeah, the HEDS module on the servomotor I sell operates at 5 V.

zougloub, 2017-05-04:
OK. Closing this issue, thanks!

closed 2017-05-04 by zougloub