ottoragam/Tarocco #3: Serial port electrical interface?

zougloub, 2017-05-04:
Hi,

The doc mentions using "just a USB-to-serial converter", but doesn't mention the voltage levels (serial can be 1.8V, 3.3V, 5V, 12V... I have 3 different kinds around me right now) I'm not familiar with the Cypress PSoC and I'd like to be on the safe side, so could you precise what kind of standard is used or an example of dongle that works?

The PSoC datasheet is stating that GPIO voltage is the VDDD voltage, which I would assume would mean that the chip is sending 5V if powered by 5V... and so a TTL dongle wouldn't work... but a TTL dongle powering the chip VLOGIC could work... again, I assume, didn't try yet.

Regards,

ottoragam, 2017-05-04:
My bad, I'll add some more info to the manual. The board works with 1.8 to 5 V. You should use only one logic voltage supply for the board, as all the lines marked as 5 V are connected together.

You're correct, TTL level serial converters should work provided they share the same power supply with the PSoC. I've tested the serial interface with some FT232 and CP2104 chips, and they work just fine.

zougloub, 2017-05-04:
OK, I wired a [CP2102 board](https://www.amazon.ca/Module-Serial-Converter-CP2102-Dupont/dp/B01GO9EYMU), and I get serial input (`$000500 000200; $000500 000210;...`).

I'm planning to use the serial interface for control. So far I'm not able to make sure that TX is working, I was assuming that `u0;` would stop the 100 Hz messages, either I'm wrong or my TX is not working.

ottoragam, 2017-05-04:
Thats right, u0; should stop the messages. Try moving the motor woth the
+/- command to confirm it gets your serial data.

<https://www.amazon.ca/Module-Serial-Converter-CP2102-Dupont/dp/B01GO9EYMU>,

<https://github.com/ottoragam/Tarocco/issues/3#issuecomment-299321357>, or mute
the thread

zougloub, 2017-05-04:
I can toggle `u1;` /  `u0;` but after entering other commands I think I can crash the firmware... leaving diagnostics on, it just stops sending.

I'll install PSoC creator and take a look at the code.

closed 2017-05-04 by zougloub

ottoragam, 2017-05-04:
Does it work again after a reboot, or is the EEPROM getting stuck?

zougloub, 2017-05-05:
It works again after a reboot.