ottoragam/Tarocco #4: Status aren't sent after some condition

zougloub, 2017-05-04:
Hi,

Driving the servo with this code (python, pyserial):
```python
#!/usr/bin/env python
# Tarocco test code

import serial

port = "/dev/ttyUSB0" # eg. "COMx"
ser = serial.Serial()
ser.baudrate = 19200
ser.port = port
ser.timeout = 2.0
ser.open()
for i in range(5):
	ser.write(b"u1;")
```

It just crashes at the 5th iteration.

ottoragam, 2017-05-04:
Could you try inserting a time delay between command transmissions?

zougloub, 2017-05-04:
Yeah, I tried to insert 0.2 s sleeps between commands but it still ends up crashing after a while.
Adding a time delay shouldn't be a solution anyway, especially with a 48 MHz MCU and a 19200 b/s serial port...

ottoragam, 2017-05-04:
Yeah, I just wanted to check if that had some effect. Let me see what else I can figure out.

zougloub, 2017-05-04:
Took a look at the code to see if I could find something for this issue, and saw something else :)

closed 2017-05-24 by ottoragam