ottoragam/Tarocco #8: Docs: tried to follow upgrade SOP but getting upload issues

zougloub, 2017-05-05:
Hi,

Trying to troubleshoot issues and reload the firmware, I'm following the SOP:

0. PSoC Creator installed, ...
1. Configure serial port 57600-8N1, select file `[...].cyacd` in the Bootloader Host
2. Wait for LED lockstep
3. Immediately load (F5)

And I get errors, like `Communication port reported 'unable to read data from the target device'`, `The flash row is not valid for the selected array`.
It could be my mistake, but it seems simple enough...

ottoragam, 2017-05-05:
It appears I left some important info out of the manua (again!). Please double click on the Bootloadable_1 component box on the TopDesign.cysch tab of the Tarocco project, in PSoC creator. That'll open a configuration window, then you can switch to the dependencies tab and adjust the path th your copy of the cncboot.hex and cncboot.elf files. They can be found in the Bootloader folder provided in this repo.

zougloub, 2017-05-05:
Hi, I had figured that out already, and *I* didn't mention it. The issue happens despite having found the .hex/.elf files and rebuilding.

ottoragam, 2017-05-05:
Could you try rebuilding the cncboot project without size optimizations,
and then build again the Tarocco project. It should use a bit more than 13
kB of flash. I believe the problem you're having is caused by
incompatibilities between the different bootloader versions.

<https://github.com/ottoragam/Tarocco/issues/8#issuecomment-299512446>, or mute
the thread

phdussud, 2017-05-08:
I am not sure your issue is due to a software problem. I encountered the exact same error and I finally tracked it down to the fact that the bootloader is active on startup for only 3 seconds. You have only 3 seconds to press the program icon on the bootloader host after the board is powered up. The only way I could successfully program the new image was to connect my board to a software controlled power supply. I clicked on the power on then quickly switched to the bootloader host and clicked on the program icon. 
I hope it helps! 
Patrick

closed 2021-03-19 by ottoragam