ottoragam/Tarocco #15: Building fresh Tarocco: Communication port reported error 'Unable to read data from the target device'.

dedekcz, 2019-10-28:
Hi, Many thanks for interesting and useful project.

I made Tarocco board by design v1.1 And now facing issue to flash firmware into board. I used Arduino as USB to serial adapter. 5V, GND and RX,TX are connected from Arduino to appreciated inputs at Tarocco. When I connecting serial adapter to Tarocoo, I didn't see any LED blinking on Tarocco. And at Bootloader Host application I got (Communication port reported error 'Unable to read data from the target device'.)

So I suppose that trouble comes because MCU has no bootloader inside. I would say that I follow You procedure from manual, with selecting all files from repo, but without success. So is other procedure how to flash it correctly?

Thanks a lot for Your help.

ottoragam, 2019-10-28:
Hi!
Newly purchased PSoCs come with no bootloader from factory. You must use a
programmer instead of a USB-serial apter first. I use a cheap kit from
Cypress. It includes a programmer a a nice PSoC5 to play with CY8CKIT-059
<https://www.cypress.com/documentation/development-kitsboards/cy8ckit-059-psoc-5lp-prototyping-kit-onboard-programmer-and>

dedekcz, 2019-11-03:
Hi @ottoragam Thanks, for clarification.

Now I got CY8CKIT-059, I separated KitProg from PSoC5. and I connected:
KitProgBoard  Tx (pin 12.7) --> Rx (Tarocco)
KitProgBoard  Rx (pin 12.6) --> Tx (Tarocco)

But when I follow manual and use Tools --> Bootloader host. I getting communication error ('Unable to read data from the target device')

so is there any step before ? Thanks for Your time.

dedekcz, 2019-11-03:
All right, it looks that light comes. I used SWD interface... :)

closed 2019-11-03 by dedekcz

ottoragam, 2019-11-04:
@dedekcz 
Glad you got it working. Do let me know if you need more help with the next steps.