ottoragam/Tarocco #17: STEP/DIR interface problems

niclasku, 2020-06-20:
Hi, I bought one of Tarocco board with the proposed motor and encoder. Configuration of the board via serial interface is working well, but I am not able to control it via the STEP/DIR interface. I tried it by generating a square wave signal with an Arduino. Can you give me an advice on that? Special initialization procedure needed?

ottoragam, 2020-06-20:
Hi
You shouldn't need anything more than the pulse train to move the motor. Can you give me a more detailed description of your setup? Try looking at the error data with Serial Port Plotter. If the encoder is working correctly, you should be able to spin the motor shaft by hand and see the changes in the error graph over time.

niclasku, 2020-06-20:
Thanks for your answer. 
If I only connect via serial (4-pin header), I can see the status messages of the board. Spinning the motor shaft by hand results in an immediate correction, but I can see the error value going up or down a little. I also tried to send commands: `+100;` results in a short spin of the motor (as expected), while `+200;` results in an error (VIN + ERR LED). Clearing the error with `e1;` works. Also set all the other values to the defaults.

If I only connect DIR, STEP, VCC, GND (6-pin header) with the Arduino, I can see VIN and ERR blinking at startup and VIN glowing after startup. Connections:
- STEP <-> Digital Pin
- DIR <-> GND
- VCC <-> +5V
- GND <-> GND

I tried different square waves as input for the STEP pin (1 kHz, 10 kHz, 30 kHz), but the motor doesn't start spinning. Also double checked with a Logic Analyzer that the signal is generated properly. If I try to spin the motor by hand, the board corrects the error immediately.

niclasku, 2020-06-20:
Nevermind, the cable was bad... Now the motor spins but the board goes into error after one or two rounds. Is this expected behavior? I think I have to tune the PID values.

ottoragam, 2020-06-20:
Yes , you need to tune the PID according to your system operating
conditions. The board will go into error if the error cannot keep up with
your pulse train. Also, remember you need to account for inertia, as the
motor cannot change speed instantaneously. You should ramp up and down the
frequency of your STEP signal, or else a sufficiently high frequency signal
will cause a following error alarm.

Let me know if you need more help with the controller!

closed 2020-07-01 by ottoragam

niclasku, 2020-07-04:
Thanks again. I think I am doing a bad job on tuning the PID values. Can you provide values that work with the suggested motor? I am trying to get it run without any load in the first place. 

Another observation is that sometimes sending for example "+100;" via UART the position error counter gets increased by 100 and the motor starts spinning, but the position error counter doesn't start decreasing after that. This results in a permanent spinning. Maybe the encoder is faulty?

ottoragam, 2020-07-04:
I don't have any particular set of values for the motor. You can start a the default P2500 I0 D0 and go from there by increasing P up until the shaft starts oscillating, then lower P (maybe 20%) and  start increasing to close in on zero error.

Regarding the motor runaway, that's definitely not right. My guess is that one of the encoder channels is missing, so the controller thinks that the motor is oscillating in place (+- one count, that's why you don't see changes in the error), but it is actually spinning. This may be again a fault on a cable or connector, so please check those first.