ottoragam/Tarocco #18: PID settings communication problem

bojfik, 2020-09-02:
https://youtu.be/Y-sH2tY315Y

ottoragam, 2020-09-02:
Hi @bojfik I'll check the video in a bit and report back.

bojfik, 2020-09-02:
> Hi @bojfik I'll check the video in a bit and report back.

OK thanks.  I will just add that the bootloader is cncboot.cydsn and the firmware is those from 2017-08-12

bojfik, 2020-09-03:
> Hi @bojfik I'll check the video in a bit and report back.

More explain on this video
https://youtu.be/0OufBB6x35c
Sorry for my bad english

bojfik, 2020-09-04:
I took another CNC PCB and reprogrammed the bootloader and uploaded the firmware sequentially.  without any changes, bootloader loads correctly, firmware too, but I still don't have access to PID settings.

ottoragam, 2020-09-05:
Hey, sorry for the lapse in communication, I was kinda busy until yesterday. I saw both videos and it appears your setup is working fine. I just think there is a bit of confusion regarding the motor tuning. Let me explain:

What you're seeing on the terminal ($00500; $00510;) is the board streaming both the following error and the motor current. In this case the error is 0 because the motor is not moving yet and the current is 500 mA. You can use a program like https://github.com/CieNTi/serial_port_plotter/releases to watch the plots in real time to help with motor tuning.

In order to tune the P, I, D coefficients, you need to use a terminal program (my choice is Realterm, as it allows you to hide the data stream from the Tarooco, but anything should work) and type something like p5000;

I suggest to try the Serial Port Plotter software while moving the motor shaft by hand to get a feeling on how the updated coefficients change the motor behaviour.

Let me know if you need more help.

bojfik, 2020-09-05:
Thank you very much for your answer.  I thought what I see in the terminal is a communication error 😁 now I understand everything.

ottoragam, 2020-09-05:
No problem! Hit me up again if something else comes up.

closed 2020-09-05 by bojfik