ottoragam/Tarocco #19: PWM switching freqency and related noise

dedekcz, 2021-03-19:
Hi,

Do You know if is possible to increase PWM switching frequency from 6 kHz to 24 kHz (or above 20 kHz) to reduce noise? In my system, I have very low encoder error (less then 3 points) but sometimes motor stay at final position, and it looks that regulation still keeping to get target and this leads to noise.

Do You have such experience?
Thanks,
Ros.

ottoragam, 2021-03-20:
Hi Ros,
It should be possible to change the PWM frequency by adjusting the clock of the Timer/Counter block in PSoC Creator that feeds the pulses to the gate driver. This will also increase the load on the microcontroller, as the PID output calculation rate is tied to the PWM frequency, but I believe it should be able to handle 20+kHz. You can play with it if you wish, or give me a pair of days so I can test this myself and post the updated firmware.

dedekcz, 2021-03-21:
Yes, I believe on your expertise, So I will wait for your FW. Many thanks.

ottoragam, 2021-03-22:
I made available a new firmware version. The pulse freq should be around 23 kHz now. Please test it and let me know your findings.

dedekcz, 2021-03-23:
Hi,

I tried to build FW same way,  but after update to Driver I lost Serial communication. For confirmation of my process, I repeat build procedure with FW (2017-08-12) again, and serial connection works there.
So it looks that new FW has some issue.

ottoragam, 2021-04-03:
Sorry for the lack of communication, I've been very busy these past days. I'll check to see what might be wrong with the new firmware and write back.

labkab, 2021-08-01:
Hi
I have the same issue with audible sound for servomotor
so flashed new ultrasonic pwm frequency firmware but does not work.
Symptom is that motor spins in one direction without command without stopping without error..
With old firmware works fine..

petetom, 2021-08-29:
Hello,
I have the same issues with the latest firmware 03-21-2021.  I noticed that after programming with this latest version both LEDs do not stop flashing after programed.  Power is cycled LEDs stop flashing after 3 sec as expected and with no pulse inputs the motor moves and not able to create any faults like disconnecting the encoder signals.  Also there is no debug info coming back from the serial port.

Have you been able to get the this version to properly work?
The only way to program is to use the no-optimization GCC-493 path for the dependency. 

Thanks,
-pete