ottoragam/Tarocco #20: pid tuning

labkab, 2021-08-14:
Hello.
Iam using tarocco driver for X axis of a trotec 8003 laser(<50w motor / 500 lines encoder) that has very minimal mechanic load on x axis (head has very low mass and all the system has low mass)
With original driver this laser has 2000mm/s rapid speed and enough acceleration to achieve this speed.
The following error was minimal with original drivers..
I have tried various values for P/I/D and seems that pid tuning is pure.
I and D values does not have much effect for fine tuning..
With P values around 2500-3000 and no I/D is the least following error that i can get
but the following error is too much  >> 1mm so does not have accuracy in high speed/acceleration
Seems that PID algorithm is pure because with original driver the accuracy on high speed/acceleration following error is much better.
Now encoder is 500 lines if i change it with 2048 i will see improvement in following error ?
Of course with low acceleration/speed following error improves but then the machine is slow..
Where in the source is the code is the that does the pid tuning ?
Thanks,

ottoragam, 2021-08-16:
Hi!
Using a higher count encoder should help with PID performance, . A 500 line encoder does sound small, especially when controlling belt driven systems like your laser, if I'm not mistaken, which would result in somewhat coarse resolution for the position feedback loop.

The PID specific code starts in the main.c file, line 227 of the 2017-08-12 firmware, specifically this part:

`        //calculate error
        error=(int16_t)ERROR_COUNTER_ReadCounter();
        //calculate change in process variable to prevent derivative kickback
        error_derivative=error-error_prev;

        //sum of errors over time
        if (motor_shutdown == 0)
            error_integral+=error;

        //limit integral error
        if(error_integral>ERROR_INTEGRAL_MAX) error_integral=ERROR_INTEGRAL_MAX;
        else if(error_integral<-ERROR_INTEGRAL_MAX) error_integral=-ERROR_INTEGRAL_MAX;
        
        error_prev=error;
        
        //output value calculation and constraining
        output=(*eeprom_kp*error)+((*eeprom_ki*error_integral)>>7)-(*eeprom_kd*error_derivative);`

I'd think you should be able to increase the P value from the 2500-3000 range you have now (without touching I and D for now). Can you tell me what happens if you do that?

labkab, 2021-08-17:
Thank you for help.
500 lines is the original encoder the servo motor has.
With that encoder each step is 10um length belts-pulleys are high quality 1.5mm teeth profile with backlash < 0.03mm.
If i use 2000 lines encoder each step will be 2.5um but with following error ~500 this is 1250um or 1.25mm 
and with ferror 250(if i reduce acceleration/speed) will be 0.625mm which is very large compared with laser spot size that is 0.1mm.
The factory servo driver had with 500 lines encoder very minimal ferror with high speeds accelerations..(2000mm/s rapids)
so i think the algorithm of factory servo driver was not simple pid but had more complex tuning algorithm
the original performance was like that https://www.youtube.com/watch?v=0luugsUN-vE

I have increased P to 3500-4000 and i get big oscillations also with 3000 get some oscillations without some good improvement in ferror 2500 has is the more stable.
The best can achieve now with acceptable ferror is 10-20mm/s that is far worse from open loop steppers..
I will get better encoders but that will not solve the problem...
I think it needs more advanced tuning algorithm with feed forward.

Also how can increase the pwm frequency to something 21-22khz because now is audible ?

ottoragam, 2021-08-25:
Hey, sorry for the late reply, I was very busy last week.
Are you sure about the step resolution? If the motor is coupled one to one to the driving pulley, the resolution should be the ratio of pitch circumference of the driven pulley and the encoder cpr (2000 in this case). A 2.5 um resolution would mean that you have a pulley with a 5mm pitch circumference, which doesn't sound right to me.

As you mention, it is possible that the original laser servo driver performed a more complex control algorithm. I personally use the Tarocco motor controllers in my CNC router. I normally don't have an issue using P values higher than 10 000, but my drive system is screw based, so it's not directly comparable.

Try increasing the I value once the oscillations start and see if you can get rid of them. Then push the P term higher up again and try to stabilize the system with the I coefficient as necessary.

I haven't been able to debug the 20+ kHz firmware as I don't have the time at the moment. Hopefully in a few weeks I'll have enough free time to check the issue with that firmware version.

labkab, 2021-08-25:
Yes 10um sure for step resolution there is 1:3 reduction and the pulley than drivers belt is 30 teeth s2m type with 2mm pitch
i have calculated that correct...
But this is not the problem because with the same encoder-motor original servo drive had much better performance..
If increase I up to 500 ferror reduced almost to half but does not have good behavior at the change of direction so i set I again to 0. D seems to do nothing..
X stage of laser has very minimal mass (25% of same type chinese lasers) so it is almost near to nothing the motor load.
I think without better error reduction algorithm can not perform better for these type applications(high acceleration/speed)
Something like FF1 of linuxcnc pid i will help a lot.

Mill has a lot less acceleration and speed than laser and much more load.

ottoragam, 2021-08-25:
I'll try to add the feed forward term next week. I'll let you know when it
is ready for testing.

lawrence-jeff, 2022-09-06:
Any update on this thread - was looking at the driver to replace the 'servos' on my laser, but if this is better suited for a slower CNC then I might look at a different solution

labkab, 2022-09-08:
Me too.
Seems dead no updates..
The algorithm for posisition loop is pure 
but iam not sure whitch commercial solution has good results.. Do you know something tested?

lawrence-jeff, 2022-09-08:
I know of one person who has converted his laser using the Gecko G-320X drivers, he reported it works great but took tuning with an oscope. These are motors with an encoder on the shaft.
Another user used them with a strip encoder (I guess what some brands use) and reported results that were middle of the road, they work but can't go nearly as fast as stock.

 I was hoping for a more affordable option as I usually do more vector based work where speed isn't the most critical item

labkab, 2022-09-08:
The tarocco hardware is ok the software is poor.
If had proper code documentation maybe it was more easy for improvements. 
smtbl servo drive has good turning algorithm  (P,PIFF,PIFF + velocity observer + motor model)
here is a comparison
https://github.com/rene-dev/stmbl/wiki/servo-drives-overview

JamesNewton, 2022-09-08:
May I point out that the software is open source? And therefore open to improvement by others? Perhaps some posts on tech friendly blogs asking for help improving the software would connect those who can improve it with those who need it improved?